edk2-porting/edk2-rk3588 #181: Netboot option

peltho, 2025-01-09:
I flashed `EDK2` on my `Radxa Rock 5C lite` **SPI flash module** successfully but now I'm wondering how I can achieve netboot from a TFTP server for instance.

There's no boot option in stock EDK2 and Shell does not provide TFTP command.
Is it somehow possible to compile it to embed network boot options?

BR

peltho, 2025-01-09:
I suppose this is achievable with `--edk2-flags` param but I doesn't seem to work

mariobalanica, 2025-01-09:
Netboot is only supported on platforms with Realtek PCIe NICs.

I would also advise against using images meant for other boards on unsupported platforms, even more so RK3582 which isn't supposed to have all the hardware blocks of a RK3588S enabled. Not that this would necessarily cause damage, but you may see the system randomly crashing due to a half-broken CPU core being enabled when it shouldn't have been.

Rippanda12, 2025-01-10:
@mariobalanica On my 5c lite it doesn't even boot with edk bc my cpu is messed up

peltho, 2025-01-10:
@mariobalanica Thanks for answering so fast!
I see, however thanks to a PoE HAT it is achievable. That's too bad it's not provided even if it wouldn't work without that HAT.
Now I'm considering using U-Boot instead unfortunately 😥

mariobalanica, 2025-01-14:
I've made some progress on this and may find time to finish the driver next week.

mariobalanica, 2025-01-22:
The latest build from here: https://github.com/edk2-porting/edk2-rk3588/actions/workflows/build.yml now has onboard Ethernet working.

I've successfully tested it on the ROCK 5A with HTTP boot.

closed 2025-02-28 by mariobalanica