edk2-porting/edk2-rk3588 #182: Orange Pi 5 Plus: No NVME On Device Tree Mode

alexander-lysenko, 2025-01-16:
I'm using Orange Pi 5 Plus 8 GB with EDK2 firmware v0.12.2 flashed to SPI NOR.
I'm trying to install a Linux system on my NVME drive (HP EX900 M.2 120GB, installed in M.2 M-Key slot) and I get it unrecognized on Device Tree Mode after live system booted. 
However it is available on ACPI mode, but I get unrecognized WiFi (Intel AX210NGW at M.2 E-Key slot) and the boot process is about 3 times slower.

Here are the options I've tried:
 * [Armbian rk3588-live-iso](https://github.com/amazingfate/rk3588-live-iso) from README (kernel 5.10-legacy)
 * [Armbian - Ubuntu 24.04 Noble](https://dl.armbian.com/uefi-arm64/Noble_current_server-kisak) (server image, kernel 6.6.60)
 * [Archboot](https://release.archboot.com/aarch64/2025.01/iso/archboot-2025.01.07-02.29-6.12.1-2-aarch64-ARCH-aarch64.iso)  (kernel 6.12.1)
 * [Ubuntu Server 22.04.5](https://cdimage.ubuntu.com/ubuntu/releases/jammy/release/ubuntu-22.04.5-live-server-arm64.iso) (official image, kernel 6.8.0-40)
 * [Ubuntu Server 24.04.1](https://cdimage.ubuntu.com/ubuntu/releases/noble/release/ubuntu-24.04.1-live-server-arm64.iso) (official image, kernel 6.8.0-41)

mariobalanica, 2025-01-16:
What does `dmesg` say? 

Also, there isn't much point in using kernels older than 6.10 as they lack a lot of things for RK3588 support.
Try something like Fedora Workstation 41 or Ubuntu 24.10.

alexander-lysenko, 2025-01-16:
> What does dmesg say?

<details>
  <summary>dmesg | grep pci</summary>

  ```
[    0.089829] platform a40c00000.pcie: Fixed dependency cycle(s) with /pcie@fe180000/legacy-interrupt-controller
[    0.090229] platform a41000000.pcie: Fixed dependency cycle(s) with /pcie@fe190000/legacy-interrupt-controller
[    0.107429] platform a40000000.pcie: Fixed dependency cycle(s) with /pcie@fe150000/legacy-interrupt-controller
[    0.107920] platform a40800000.pcie: Fixed dependency cycle(s) with /pcie@fe170000/legacy-interrupt-controller
[    3.263316] rockchip-dw-pcie a40000000.pcie: host bridge /pcie@fe150000 ranges:
[    3.265787] rockchip-dw-pcie a40000000.pcie:       IO 0x00f0100000..0x00f01fffff -> 0x00f0100000
[    3.269410] rockchip-dw-pcie a40000000.pcie:      MEM 0x00f0200000..0x00f0ffffff -> 0x00f0200000
[    3.272102] rockchip-dw-pcie a40000000.pcie:      MEM 0x0900000000..0x093fffffff -> 0x0040000000
[    3.275009] rockchip-dw-pcie a40000000.pcie: iATU: unroll T, 8 ob, 8 ib, align 64K, limit 8G
[    4.341614] rockchip-dw-pcie a40000000.pcie: Phy link never came up
[    4.344203] rockchip-dw-pcie a40000000.pcie: PCI host bridge to bus 0000:00
[    4.346912] pci_bus 0000:00: root bus resource [bus 00-0f]
[    4.348990] pci_bus 0000:00: root bus resource [io  0x0000-0xfffff] (bus address [0xf0100000-0xf01fffff])
[    4.351387] pci_bus 0000:00: root bus resource [mem 0xf0200000-0xf0ffffff]
[    4.353540] pci_bus 0000:00: root bus resource [mem 0x900000000-0x93fffffff] (bus address [0x40000000-0x7fffffff])
[    4.356038] pci 0000:00:00.0: [1d87:3588] type 01 class 0x060400 PCIe Root Port
[    4.358213] pci 0000:00:00.0: BAR 0 [mem 0x00000000-0x3fffffff]
[    4.360285] pci 0000:00:00.0: BAR 1 [mem 0x00000000-0x3fffffff]
[    4.362308] pci 0000:00:00.0: ROM [mem 0x00000000-0x0000ffff pref]
[    4.364376] pci 0000:00:00.0: PCI bridge to [bus 01-ff]
[    4.366334] pci 0000:00:00.0:   bridge window [io  0x0000-0x0fff]
[    4.368325] pci 0000:00:00.0:   bridge window [mem 0x00000000-0x000fffff]
[    4.370346] pci 0000:00:00.0:   bridge window [mem 0x00000000-0x000fffff 64bit pref]
[    4.372547] pci 0000:00:00.0: supports D1 D2
[    4.374346] pci 0000:00:00.0: PME# supported from D0 D1 D3hot
[    4.381812] pci_bus 0000:01: busn_res: can not insert [bus 01-ff] under [bus 00-0f] (conflicts with (null) [bus 00-0f])
[    4.384499] pci 0000:00:00.0: BAR 0 [mem 0x900000000-0x93fffffff]: assigned
[    4.386540] pci 0000:00:00.0: BAR 1 [mem size 0x40000000]: can't assign; no space
[    4.388561] pci 0000:00:00.0: BAR 1 [mem size 0x40000000]: failed to assign
[    4.392506] pci 0000:00:00.0: ROM [mem 0xf0200000-0xf020ffff pref]: assigned
[    4.394621] pci 0000:00:00.0: PCI bridge to [bus 01-ff]
[    4.396468] pci_bus 0000:00: Some PCI device resources are unassigned, try booting with pci=realloc
[    4.398680] pci_bus 0000:00: resource 4 [io  0x0000-0xfffff]
[    4.400577] pci_bus 0000:00: resource 5 [mem 0xf0200000-0xf0ffffff]
[    4.402500] pci_bus 0000:00: resource 6 [mem 0x900000000-0x93fffffff]
[    4.406654] pcieport 0000:00:00.0: PME: Signaling with IRQ 85
[    4.410081] pcieport 0000:00:00.0: AER: enabled with IRQ 86
[    4.464973] rockchip-dw-pcie a40800000.pcie: host bridge /pcie@fe170000 ranges:
[    4.467147] rockchip-dw-pcie a40800000.pcie:       IO 0x00f2100000..0x00f21fffff -> 0x00f2100000
[    4.470760] rockchip-dw-pcie a40800000.pcie:      MEM 0x00f2200000..0x00f2ffffff -> 0x00f2200000
[    4.473036] rockchip-dw-pcie a40800000.pcie:      MEM 0x0980000000..0x09bfffffff -> 0x0040000000
[    4.475445] rockchip-dw-pcie a40800000.pcie: iATU: unroll T, 8 ob, 8 ib, align 64K, limit 8G
[    4.677626] rockchip-dw-pcie a40800000.pcie: PCIe Gen.2 x1 link up
[    4.679752] rockchip-dw-pcie a40800000.pcie: PCI host bridge to bus 0002:20
[    4.683768] pci_bus 0002:20: root bus resource [bus 20-2f]
[    4.685683] pci_bus 0002:20: root bus resource [io  0x100000-0x1fffff] (bus address [0xf2100000-0xf21fffff])
[    4.690425] pci_bus 0002:20: root bus resource [mem 0xf2200000-0xf2ffffff]
[    4.694737] pci_bus 0002:20: root bus resource [mem 0x980000000-0x9bfffffff] (bus address [0x40000000-0x7fffffff])
[    4.697138] pci 0002:20:00.0: [1d87:3588] type 01 class 0x060400 PCIe Root Port
[    4.699168] pci 0002:20:00.0: ROM [mem 0x00000000-0x0000ffff pref]
[    4.701060] pci 0002:20:00.0: PCI bridge to [bus 01-ff]
[    4.702854] pci 0002:20:00.0:   bridge window [io  0x0000-0x0fff]
[    4.704720] pci 0002:20:00.0:   bridge window [mem 0x00000000-0x000fffff]
[    4.706637] pci 0002:20:00.0:   bridge window [mem 0x00000000-0x000fffff 64bit pref]
[    4.708821] pci 0002:20:00.0: supports D1 D2
[    4.710738] pci 0002:20:00.0: PME# supported from D0 D1 D3hot
[    4.718133] pci 0002:20:00.0: Primary bus is hard wired to 0
[    4.720092] pci 0002:20:00.0: bridge configuration invalid ([bus 01-ff]), reconfiguring
[    4.723940] pci 0002:21:00.0: [8086:2725] type 00 class 0x028000 PCIe Endpoint
[    4.726203] pci 0002:21:00.0: BAR 0 [mem 0x00000000-0x00003fff 64bit]
[    4.728688] pci 0002:21:00.0: PME# supported from D0 D3hot D3cold
[    4.733737] pci_bus 0002:21: busn_res: [bus 21-2f] end is updated to 21
[    4.737760] pci 0002:20:00.0: bridge window [mem 0xf2200000-0xf22fffff]: assigned
[    4.739941] pci 0002:20:00.0: ROM [mem 0xf2300000-0xf230ffff pref]: assigned
[... 79 lines cut ...]
[    5.415985] pci 0004:41:00.0: PME# supported from D0 D1 D2 D3hot D3cold
[... cut ...]
[    5.436469] pci_bus 0004:41: resource 0 [io  0x300000-0x300fff]
[    5.437598] pci_bus 0004:41: resource 1 [mem 0xf4200000-0xf42fffff]
[    5.440002] pcieport 0004:40:00.0: PME: Signaling with IRQ 124
[    5.442319] pcieport 0004:40:00.0: AER: enabled with IRQ 125
[   17.337029] rfkill_gpio rfkill: rfkill-pcie-wlan device registered.
```
</details>

and here is full output: [dmesg.txt](https://github.com/user-attachments/files/18445929/dmesg.txt)

Taken from Ubuntu 24.10 with kernel 6.11.0-8

ksalaba, 2025-02-28:
@alexander-lysenko  I was able to get my NVME to workby using ACPI. I could not get it to work with Device Tree, or Both mode.

amazingfate, 2025-02-28:
I just built debian iso with mainline 6.14-rc4 kernel, you can have a try: https://github.com/amazingfate/rk3588-live-iso/releases/tag/20250227-1505

I'm going to drop the iso with old 5.10 legacy kernel, and if grub can support dt overlay in the future, I will build iso with 6.1 vendor kernel.

alexander-lysenko, 2025-03-07:
I did get an OEM SSD by Micron and it works great onboard. I also tried WD Blue SM570 taken from my main PC, there were no problems to recognize it with images I provided on the topic.

It turned out that my HP EX900 SSD was just incompatible with this board. I'm not sure that is an EDK2 issue - I've tried the SSD the vendor's bootloader and it didn't work at all.

mariobalanica, 2025-03-07:
Does the EX900 show up in the UEFI boot menu at least?

closed 2025-03-15 by mariobalanica

yodaldevoid, 2025-07-26:
I had a similar issue with a Patriot P300 in a FriendlyELEC NanoPi R6C. The drive would show up in the UEFI boot menu no matter the settings. If in device tree mode the initramfs of Linux would load, but at that point the the drive would not show up for Linux and booting would halt as it failed to find the rootfs. In ACPI mode the drive worked fine, though there were plenty of other devices that didn't show up. Swapping out for a Micron drive that I had on hand "solved" the issue.

What is strange is that the Patriot drive would very rarely work on device tree mode, but I couldn't find a pattern. I'm guessing there is a race condition somewhere that the Patriot drive triggers but the Micron drive does not.