edk2-porting/edk2-rk3588 #184: Hardware acceleration

awesomefroggy29, 2025-01-22:
> Use proper device tree pair with your kernel. Mainline Mali GPU driver doesn't support acpi mode yet. So it is not a uefi issue. 

 _Originally posted by @matalama80td3l in [#183](https://github.com/edk2-porting/edk2-rk3588/issues/183#issuecomment-2603599769)_

how do I use a proper device tree pair? and I don't need acpi mode.

mariobalanica, 2025-01-22:
As previously mentioned, this is not an UEFI issue, so I'll have to close it. But to answer your question: the firmware already provides mainline-compatible DTBs. All you have to do is update your kernel to 6.13 or newer.

Please keep in mind that mainline Linux support is still experimental. Any issues encountered there are outside this project's scope.

closed 2025-01-22 by mariobalanica

awesomefroggy29, 2025-01-22:
i am trying to use windows with hardware acceleration

awesomefroggy29, 2025-01-25:
oh i understand, windows has a whole different way that is uses the gpu vs linux. thanks!