edk2-porting/edk2-rk3588 #187: (repost) Windows BSoD on boot

awesomefroggy29, 2025-02-08:
I was trying to install windows (which has worked before) and I wanted to start from scratch, so I deleted the windows/microsoft boot manager from the boot options, and now every time I try to boot windows, it gives me a spinning loading logo for 30 seconds, then blue screens saying “inaccessible boot device”

Is there any way to fix this?

mariobalanica, 2025-02-08:
There's probably something wrong with your existing installation, unfortunately I don't have any good suggestions other than performing a reinstall from scratch.

Please note that this is the bug tracker for UEFI firmware-realted issues. While Windows is one of the OSes that can boot with this firmware, we no longer offer any kind of support for it.

closed 2025-02-08 by mariobalanica

awesomefroggy29, 2025-02-09:
I got it to work, just had to use an sd card instead of spinor