edk2-porting/edk2-rk3588 #193: The monitor LG (1920*1080 60) is not working

Gluk41, 2025-03-23:
Ubuntu 24.10 kernel 6.13.7 (full update & Oibaf drivers install) && Fedora workstation mainline kernel (full update)
Force UEFI GOP Display enabled - work fine
Force UEFI GOP Display disable - not working
Message:
[image]
[image]
It works fine at the same time gnome remote desktop (without hardware acceleration)
How do I make the monitor work?

mariobalanica, 2025-03-26:
Does the monitor work in UEFI at its native resolution, or is it limited to 640x480?

Kernel logs suggest that the monitor does not respond to EDID read commands for some reason. What happens if you boot without the monitor connected and plug it in after a couple of seconds?

> You can only see the picture using Gnome Remote Display or connect it to a TV via hdmi

So if I understand correctly, a TV works fine but not this particular monitor?

Gluk41, 2025-03-26:
> Does the monitor work in UEFI at its native resolution, or is it limited to 640x480?

works fine with native resolution

> Kernel logs suggest that the monitor does not respond to EDID read commands for some reason. What happens if you boot without the monitor connected and plug it in after a couple of seconds? 

It will show what is on the screenshots. That's the only way I could get anything out of him.
That's what's in the first two screenshots. 
If you leave it on, the monitor just freezes and does not show anything. At the same time, you can connect via RDP. I did that to show screenshots and logs.

> So if I understand correctly, a TV works fine but not this particular monitor?

That's not quite true. If you just plug in and boot up, the TV doesn't work. 
If you connect the TV after the system is fully loaded, it works fine.

mariobalanica, 2025-03-26:
Okay, so both displays work fine in UEFI.

> It will show what is on the screenshots. That's the only way I could get anything out of him.
That's what's in the first two screenshots.

If you plug the monitor in after Linux has booted, it displays that "out of range" message?

> That's not quite true. If you just plug in and boot up, the TV doesn't work.
At the same time, the Efi menu is working fine.
If you connect the TV after the system is fully loaded, it works fine.

So the TV only works in Linux if you plug it in after boot, correct? What exactly happens when you boot with it connected? Anything interesting in kernel dmesg?

mariobalanica, 2025-04-01:
HDMI compatibility issues are expected with kernel 6.13. They have been mostly fixed in the upcoming 6.15 release. Closing this one since it's not firmware related.