edk2-porting/edk2-rk3588 #224: Radxa Rock 5 ITX+ support

jasonpanosso, 2025-08-02:
I see that the Radxa Rock 5 ITX (non-plus) has platinum-tier support, is this true for the newer ITX+ as well?

VennStone, 2025-08-08:
I copied it to SPI Flash and things seem to be working. Installed Fedora and Ubuntu.

nmaas87, 2025-08-08:
@VennStone That sounds interesting, with which command did you do that?
````rkdeveloptool db imagefile````?

VennStone, 2025-08-08:
@nmaas87 I used Gnome disk utility.

nmaas87, 2025-08-09:
@VennStone Thanks for the hint, that helped.
@jasonpanosso / @VennStone - can confirm it works. After I erased the eMMC ( https://docs.radxa.com/en/rock5/rock5itx/low-level-dev/maskrom/erase - needed, otherwise it will never boot something from the SDcard! ), I used the https://www.armbian.com/radxa-rock-5-itx/ Armbian 25.2.2 Noble Gnome release, put it onto an SD Card and booted from it. Within Ubuntu, I opened the System Setup, Disk Utils, formated the SPI Flash (mtdblock0) and restored the mtdblock0 via the rock-5-itx_UEFI_Release_v1.1.img / UEFI image. Image transported via USB stick to the board. I shutdown the board, unplugged the SDcard and it worked.

The errors I saw were following:
- Not booting from the SDcard - there was something on the eMMC, this needs to be wiped first
- Not reacting to Keybaord Inputs: I used it via Sipeed NanoKVM and this did not work if the NanoKVM was NOT in the HID-only mode, which is the default. In the non-HID-only mode, it does show the NanoKVM as ethernet interface, massstorage device and as keyboard/mouse, which does annoy the UEFI. In HID-only mode (which can be configured in options of NanoKVM) it is strictly Mouse and Keyboard - and this works
- The error upon boot shown in the picture is not an UEFI error itself, it tried to boot from an USB stick I used for data transfer and accidentally still had x64 Ventoy on it.

I saw that VMware ESXi 8.x did boot successfully and I will see what else is possible.

nmaas87, 2025-08-09:
Errors found so far: PWM control of the CPU fan is not possible, can only switch on or off, the duty cycle does not have any effect, it will always run on full blast until an OS takes over. Probably it does need some changes to fully support ITX+ variant.

nmaas87, 2025-08-10:
Next: There seems to be no option to get it e.g. booting 6.1.xxx Armbian from NVMe on the ITX+, no matter what I do. Booting from eMMC is working, but not from NVMe.

puleglot, 2025-08-31:
I'm running Gentoo with 6.12.* kernel on Rock 5 ITX+ with [M.2 to Hexa SATA Adapter](https://radxa.com/products/accessories/m2-to-hexa-sata-adapter). Gentoo is installed on SATA SSD drives. The only problem I have is that `efibootmgr` creates boot entries without PcieRoot(...)/Pci(...)/Pci(0x0,0x0)/Sata(...) components which is not supported by the firmware. So I have to create them from the firmware configuration menu.

puleglot, 2025-08-31:
> The only problem I have is that `efibootmgr` creates boot entries without PcieRoot(...)/Pci(...)/Pci(...)/Sata(...) components which is not supported by the firmware. So I have to create them from the firmware configuration menu.

And `efibootmgr --full-dev-path` doesn't work